HALRobotics/Beta #115: HAL Not Opening in Grasshopper

acation, 2021-09-22:
### Description

I have tried installing and uninstalling HAL several times, and it will not load in Grasshopper. I have tried opening HAL in Rhino 5, 6, and 7. None of these will work as expected. When opening Grasshopper I get the following error in the Rhino command line:

HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'.

Upon installation there are no apparent errors:
[image]
However, when I open Grasshopper I do not receive a prompt to login to HAL, and HAL does not show up in my GH toolbar. In my Libraries folder, there is a link file, presumably that will link to the ProgramData folder which appears to contain the correct files

[image]

[image]

Here is the installation log:
[Installation.log](https://github.com/HALRobotics/Beta/files/7207588/Installation.log)

### Steps to Reproduce

1. Install HAL through the HAL Framework .exe
2. Open Rhino/Grasshopper

**Expected behavior:** HAL loads properly

**Actual behavior:** HAL does not load at all

**Reproduces how often:** 100%

### Versions

HAL Version 1.4.6-beta
Windows 10 
Version 20H2 
OS Build 19042.1237

acation, 2021-09-22:
Sorry, was a duplicate, and realized this was a compatibility issue with Enscape

closed 2021-09-22 by acation

tristangobin, 2021-09-22:
Glad you found it anayway. Enjoy programming robots !