HALRobotics/Beta #117: [ERROR] INCOMPATIBLE CONTROLLER. THIS CONTROLLER CANNOT EXPORT PROCEDURES

franc1994, 2021-10-12:
I have been using licensed Hal Robotics for about 6 months. This morning it is giving me an unusual problem. When I try to export the RAPID code of any geometry through the "export procedures" component to which I connect controller, solution, panel for destination and set true, the error appears with the following statement:

[ERROR] Incompatible Controller. This controller cannot export procedures

I use an ABB 6700 robot 3.2 m 150 kg
I have tried changing controllers, using generic, IRC5 compact, IRC5 compact V2, IRC5 dual cabinet, IRC5 single cabinet but nothing has been solved.

Can you help me?

Translated with www.DeepL.com/Translator (free version)

sebandraos, 2021-10-12:
Hi @franc1994 ,
A few questions so we can start trying to get you back up and running:
1. Which versions of Rhino, Grasshopper, the HAL Robotics Framework and Windows are you running? 
2. Did you run an update to the HAL Robotics Framework this morning? 
3. Have you updated Windows recently?
4. Do you have a broken file that you can share with us?
5. Have you configured the robot controller in any particular way?
6. Is this occurring in new files or existing files or both?
7. Are you using a personal device or something that might have been reconfigured overnight by some system admin?
7. Is there anything else that might be useful for us to know about configuration changes to your system?
Thanks

thibaultschwartz, 2021-10-12:
Try exactly
C:\Rapid Code
as suggested by @sebandraos 

it is NOT equivalent to WHAT YOU POSTED ABOVE:
C:Rapid Code
"C: Rapid Code"
"C:/Rapid Code"

When you are not sure, navigate to your folder, then press the Shift key, right-click in the window, and choose Copy as Path.

franc1994, 2021-10-12:
> Hi @franc1994 , This is working absolutely fine on my side so there is clearly something different about the setup or file that you have that we can't discern from a few screenshots. Can you share any of the files that aren't working so we can take a closer look? Or can you create a new basic file with the same issue so we can try and debug that for you?
> 
> Thanks

[Basic File.zip](https://github.com/HALRobotics/Beta/files/7329439/Basic.File.zip)

Let me know if you are able to open it

thibaultschwartz, 2021-10-12:
I just tried your file and it's working as expected, so I think something went wrong during the update, which corrupted the installation.

Could you please:
- send us the Installation.log file located in C:\ProgramData\HAL, 
- then proceed with a complete uninstall https://docs.hal-robotics.com/Grasshopper/0-Administration-and-Setup/Contents.html#05-complete-uninstallation
- reinstall using https://halinstaller.blob.core.windows.net/windows/HALRoboticsFramework.exe
- retry opening your file

franc1994, 2021-10-12:
[Installation.log](https://github.com/HALRobotics/Beta/files/7329553/Installation.log)

> I just tried your file and it's working as expected, so I think something went wrong during the update, which corrupted the installation.
>
> Could you please:
>
> * send us the Installation.log file [...]

franc1994, 2021-10-12:
I did, and now i got new error when i open grasshopper. HAL label is not there anymore
[image]

franc1994, 2021-10-12:
The first thing that comes is the window in the last picture I sent you. The I click "close" and grasshopper runs (without the label HAL in the upper part) and this is the command history:
[image]

thibaultschwartz, 2021-10-12:
fixed with a clean reinstall