HALRobotics/Beta #119: TRASFORM TARGET PROBLEM

franc1994, 2022-02-07:
Hi everyone,

I have a big problem and I hope you can help me:

- the robot should follow the black path you see in the image [1], it is more or less a chair;
- the problem occurs in the arcs of which it is possible to see the axes for each point in the image. In the one on the left I transformed the target as tangent to curve and it works, for the arc on the left I did the same and identical thing, but it doesn't work. I have no idea why honestly.
-
Can you help me? Thank you!

[image]

thibaultschwartz, 2022-02-07:
Can you upload a Rhino file containing only the curves, and a GH file containing only the curve parameters, target and transform target? Please note that yout can also use the second overload of Target From Curve to achieve the same result in one go: 
[image]

franc1994, 2022-02-07:
Yes, here you are the files:

[TRANSFORM TARGET ERROR.zip](https://github.com/HALRobotics/Beta/files/8016550/TRANSFORM.TARGET.ERROR.zip)

thibaultschwartz, 2022-02-07:
Looking at your file, it seems that there is indeed a bug in the tangent component, that is only visible when a reference input is given, so thanks for reporting this. I will fix it for the next update, in the meantime you can use one of the options in this file:
[TemporaryTangentFix.zip](https://github.com/HALRobotics/Beta/files/8018391/TemporaryTangentFix.zip)

thibaultschwartz, 2022-02-08:
I am closing this as fixed, please reopen if it still does not work on your side.

closed 2022-02-08 by thibaultschwartz

sebandraos, 2022-02-08:
This fix will be included in our next update