HALRobotics/Beta #123: Can't excute solution with solving resolution Targets only

fonstein, 2022-03-28:
### Prerequisites

* [X ] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description
It is not possible to execute solution with a curve target where the solving resolution is set to targets only. The same solution is solved, and can be exported. 

### Steps to Reproduce
1. Create a curve/polyline in Rhino and load in Grasshopper
2. Feed curve into Target component where "from curve" is chosen
3. Feed Target into Move component. Cartesian space. 
4. Robot cell configured with KUKA KR60 LR HA C, HSD ES929 Spindle.
5. Feed the procedure into Controller together with Mechanism mentioned in previous step. 
6. Feed controller into Solver. 
7. Chose Solver settings "Targets only"
8. Connect Solution to Execution component for simulation and to Export component to export. 

**Expected behavior:** 
It is expected that a solution is calculated and that the solution can be executed and exported.

**Actual behavior:** 
A solution is solved/calculated, and it can be exported. It can not be executed with the execute component. When the play button is clicked, nothing happens. 
If the "Targets only" setting is changed to e.g. "Ultra fine", it is possible to execute the solution. 

**Reproduces how often:**
Always when "Targets only" setting is chosen. Tested with both simple and more complex curves.

### Versions
Windows 10. 
Version	10.0.19044 Build 19044

HAL Robotics 1.4.12-beta
Installed HAL.McNeel.V5 1.4.32-beta
HAL.ABB, HAL.FANUC, HAL.KUKA 1.4.28-beta, HAL.UniversalRobots

### Additional Information
Nothing.

thibaultschwartz, 2022-03-28:
Hi @fonstein, could you please provide a grasshopper file to reproduce the bug?
Thanks in advance,
Thibault

thibaultschwartz, 2022-03-28:
Hi @fonstein,
The "Targets only" mode is designed to bypass the simulation, and to generate exportable code directly.
I see that this option could be better documented so I will add some information on the parameter doc.

We could eventually add a simplistic simulation in target-only mode, we will discuss this feature internally.
In the meantime, you can use a low-precision solution to generate an executable and exportable simulation ("Coarse" setting).

closed 2022-03-28 by thibaultschwartz