HALRobotics/Beta #127: 1. [WARNING] Controller Instantiation Failed. No preset is selected.

krittikawalia, 2023-02-15:
Hello, 

I am trying to preset the Robot and the Controller, but even after that I am having a problen with Instantiation with both the robot and the controller. Surprisingly, I have been using HAL for more than 3 months now and I never encountered this issue, and today suddenly I got this warning. 
I am sorry, I couldn't find this problem elsewhere, could you please guide me ? 
I would be really grateful.

Thankyou very much

sebandraos, 2023-02-15:
Hi @krittikawalia , We recently migrated the catalogues to a new cloud-host. This was fixed a few updates ago so please try the following:
1. Close Rhino
2. Update the software by re-running the installer and applying updates.
3. Navigate to "%localappdata%\HAL" and delete the contents on the Mechanisms and Controllers folders.
4. Relaunch Rhino and hopefully that's it.

Please let us know if that doesn't work but that should get you back up and running.

closed 2023-02-15 by sebandraos

krittikawalia, 2023-06-23:
Hello 

I am writing to ask for a fix to the problem I am encountering with Solve Component. 
It says : 1. Method not found : '!0[] MathNet.Numerics.LinearAlgebra.Storage.VectorStorage`1.ToArray()'. 

I already tried the reinstallation, but it changed nothing. 

Could you please advise how to fix it? 

I'd be highly grateful to you. 

Wishing you a great days ahead 

Sincerely 
Krittika WALIA 

Closed #127 as completed.