HALRobotics/Beta #133: [OTHER] Issue with robot Reference declaration (wobj) in exported RAPID procedure

ldemt, 2023-12-13:
### Description

Dear HAL Team,

I am working on a robot cell with multiple robots, which are not placed on the rhino world XY plane.

In the joint example file, I declared one of the robots (which is attached on a track), and separately declared its own robot Reference coordinate system (the base of the robot), different than the rhino world XY plane. I can successfully import planes (ABB robtargets) from the robot and reorient them in the rhino World coordinate system using the robot Reference.

However, when I export a RAPID code, I have a wobj declared, which is the Robot Reference coordinate system but relative to the Rhino world (xyz : 0,0,0, quaternion 0.7,0,0,0.7). For the program to work, I need to have no wobj (or wobj0) declared, because from the point of view of the robot, its own reference coordinate system is the world (xyz : 0,0,0, quaternion 1,0,0,0).

Do you have any ideas how I could achieve the good declaration of the robot Reference so that I will get a RAPID code without such misoriented wobj ?

Best regards
[DemoFile_ErrorWithWobjV2.zip](https://github.com/HALRobotics/Beta/files/13661228/DemoFile_ErrorWithWobjV2.zip)

sebandraos, 2023-12-13:
Hi @ldemt ,

There are two solutions here:
1. Our assumption is always that in multi-robot systems that the robot controllers are aware of their global positions and therefore that the WObj/Reference declarations can be made in the Rhino world. One option, therefore is to ensure that the robots controllers have their global positions set. This will give you the added ability to share references etc. within the controllers.
2. You can force the exporter to skip declaration of the Reference/WObj with some overrides in the naming of your References. [Here is a full tutorial on the syntax for that](https://docs.hal-robotics.com/Grasshopper/6-Control/Contents.html#64-reuse-controller-data) but if you name your Reference `MyReference @! wobj0` then we'll export it as wobj0 and it should hack past your issues. If the exporter takes issue with exporting wobj0 (which it might because it's a reserved word in RAPID) then you could create another WObj in the system module with a default pose and use that instead of wobj0 in your Reference's Alias.

Hopefully one of those solves your issue (please close the Issue if it does) but if not please let us know and we'll try and find another solution.

Thanks

ldemt, 2023-12-13:
Thank you Sebastian, I did the second solution you proposed and it worked.

closed 2023-12-13 by ldemt