HALRobotics/Beta #134: Read Digital Input

avifcohen, 2023-12-27:
### Description

Hello, 
I'm working with Kuka Kr 50, 
I would like to read a digital input as a condition to set other digital and analog outputs.
Since there is no component of reading digital/analog input I would like to know if there is any recommendation or solution to solve it. 
My intention for workflow is
Read Digital Input 1 - (wait till true) 
than 
Set Digital Output 2, Set Analog Output 2. 
Thank you
Avi.

tristangobin, 2023-12-27:
You can use a custom action and directly code it and add it into your procedure.

```
WAIT FOR $IN[10]==TRUE 
```
Where IN[10] is your digital input index 10.

Please see our documentation about custom action [here](https://docs.hal-robotics.com/Grasshopper/5-Advanced-Programming/Contents.html#55-custom-actions).

tristangobin, 2023-12-27:
Glad it helped. 
I close this issue but feel free to open it.

closed 2023-12-27 by tristangobin

avifcohen, 2023-12-27:
Hi tristan, The component wait of my HAL version not include Wait signal only time and sync. my version is 1.5.4 
is there is something I missing? 
thank you

tristangobin, 2023-12-27:
Ah. sorry, we did not release this yet (it was in my debug session). In that case, I suggest you use the custom action instead.