HALRobotics/Beta #135: HAL Menu not showing up in GH interface / Rhino8

PhilKMT, 2024-02-17:
### Description

Hello HAL team,
I just migrate to Rhino 8 and un/re-install HAL . 
But the HAL menu is not showing up when openning a gh script.
I find a way to have HAL properly displayed in gh, is to launch Rhino8 with the following parameters :
"C:\Program Files\Rhino 8\System\Rhino.exe" /netfx
and than open then launch GG and then open a gh script...
Any idea of what could be the fix ?
Thanks - P.

sebandraos, 2024-02-17:
We are still working on support for Rhino 8. There are a few minor issues we don't quite have resolved but will get an update out asap. We will update here once Rhino 8 is officially supported.

Double-clicking on a Grasshopper file has never worked with our Framework because the loading sequence of Grasshopper, Rhino and our plug-in get misaligned so Grasshopper thinks our components are missing.

PhilKMT, 2024-02-17:
Hello Sebastian,

Thanks for your quick reply , so i’ll wait !

Double-clicking on a Grasshopper file works fine on my pc, openning rhino7+Gh+Hal environment+ the script

sebandraos, 2024-03-08:
We have made some progress here and the latest updates should work with Rhino 8 when it's running in .NET Framework mode. You can swap to .NET Framework (a.k.a. netfx) mode by following these instructions from McNeel - https://developer.rhino3d.com/guides/rhinocommon/moving-to-dotnet-7/#choosing-the-net-runtime-on-windows

Although all of our code is already compatible with .NET 7, we are unlikely to distribute it in that format because it will break backwards compatibility with Rhinos 6 and 7 which we still want to support for the time being. I'm therefore going to close this but please re-open if there are any further issues.

closed 2024-03-08 by sebandraos