HALRobotics/Beta #136: [BUG] <title> Hal doesn't create a SRC file

avifcohen, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

Hello, 
I'm facing a bug after the solver running the exporting to file (exporting procedure) doesn't create an SRC file. 
The component creates the folder but it's empty. 
Thank
Avi

### Steps To Reproduce

Happens after solver is running the

sebandraos, 2024-03-11:
Hi Avi,
Can you please let us know which of our softwares and extensions you're using as well as which version of Rhino.
You can get the exact version of the Framework and all installed extensions in the HAL -> About menu or via the installer. Please also include the OS and what version of the OS you're running.
Thanks

sebandraos, 2024-03-11:
I have just checked our automated testing and everything is working as expected and this file is working for me, so I assume this is related to something more specific than just the export.

[KUKA_ExportTest.zip](https://github.com/HALRobotics/Beta/files/14559308/KUKA_ExportTest.zip)

avifcohen, 2024-03-12:
[image]
Thank Sebastian I will check your file.
I attach my Hal version details. I'm running it on Rhino 7. 
I also have to say that this thing started to happen to me since i updated yesterday my license. 
the GH file if you want to have a  look is in the following link (due to size limitation of github)
https://we.tl/t-MQoaGgOH8T
Thanks again
Avi

sebandraos, 2024-03-15:
Hi Avi, 
Did you manage to export with our test file?

avifcohen, 2024-03-18:
Hi Sebastian, yes I did.
I found that it takes around15 min to generate the script from my GH file.. which is a bit weird in compare to amount of data. I used to export things for UR5e and Kuka before that took much shorter time.  Anyway, it seems that its just super heavy. 
BTW is there any update for Rhino 8?

thibaultschwartz, 2024-03-18:
Hi Avi,
15 minutes is indeed a long time, so we will have to investigate. I suppose you are trying to export a big trajectory with thousands of targets?

sebandraos, 2024-03-18:
[image]

Look out for an update to HAL.KUKA coming in the next couple of days. This is your test file and before the update this was taking 13 mins on our testbed so hopefully the difference should be similar on your end.

sebandraos, 2024-03-18:
And for Rhino 8 - please see https://github.com/HALRobotics/Beta/issues/135

sebandraos, 2024-05-20:
Closing as stale. Please reopen if necessary

closed 2024-05-20 by sebandraos