HALRobotics/Beta #137: [BUG] <Issues of Connecting the HAL account>

Jackcrabb, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

When I start the grasshopper and it shows the log-in window of HAL, but I cannot log in my account, it tells me that I need to check my internet connect but my Internet still working.
[image]

### Steps To Reproduce

[image]
[image]

sebandraos, 2024-03-14:
Hi Jack, 
Everything looks like it's supposed to be working from our end so a few quick follow-ups:
1. Are you 100% sure your password is right?
2. Can you log in to the [user portal online](https://user.hal-robotics.com) with the same credentials?
3. Have you ever managed to connect from that PC?
4. Can you get to this web page from your PC? https://user.hal-robotics.com/api/token (it should just say "Bad Request.")
Thanks

Jackcrabb, 2024-03-15:
1. Yes, I am very sure about that.
2.Yes, I can
3.No, Its my first time to connect from this PC.
4. Yes, I can
Thanks for reply

sebandraos, 2024-03-15:
That's odd then. Can you try to log in on another PC? 
If so, then it's possible that there is a plugin on the new PC that's creating a conflict.

sebandraos, 2024-05-20:
Closing as stale. Please reopen if necessary

closed 2024-05-20 by sebandraos

ldemt, 2024-07-02:
Hello dear HAL Team,

I faced the same issue on one of the PCs I use with HAL robotics.
This PC is connected through ethernet cable to internet, and also, with a second network card and ethernet cable, to my robot controllers network.

When I disconnect the robot ethernet cable while logging in, it works.

Best regards

sebandraos, 2024-07-02:
Hi @ldemt ,

Thanks for working that out and letting us know. That gives us a good starting point for finding a definitive fix for this issue. By that logic it almost sounds like a Windows networking issue but we'll try and reproduce it here in the same circumstances and see if that helps.

We have also added a Licensing.log file to %ProgramData%\HAL which can be sent to the-team@ our normal domain hal-robotics.com referring to this issue which should help us dig even more.

Thanks again