HALRobotics/Beta #140: Dynamic Input for Solve

DogaKiralioglu, 2024-05-17:
### Description

Hello everyone! Thank you for the plug-in, it is pretty amazing. I wanted to ask about dynamic/ iterative inputs for the Solve node.

I am working on a project that requires machine vision to control robot motions. I was able to put together a Grasshopper script where the robot repeats a hand traced path, but I was only able to do so after cacheing the points on the path. Would there be a way to update the inputs during the simulation, so that the robot is responsive? 

I have seen that there is a Create a Loop feature coming soon, so I was hoping maybe there is already some movement on this topic as well. I would also love to contribute if there is anything I can do.

Thank you in advance!

https://github.com/HALRobotics/Beta/assets/168581816/a37974be-d71a-45a8-a17b-b20131e4020a

Best, Doga

thibaultschwartz, 2024-05-20:
Hi Doga,
Although the HAL Robotics Framework can be used to stream live positions to a selection of robot brands, including UR, we decided not to publish this feature in the Grasshopper SDK yet, mostly for safety reasons. You can still use sensors and robots in a sequential way, as you found out, by recording sensor data to generate a robot program, but I don't think teleoperation features will be made public in the near future.
If your project is commercial and requires custom development, we can discuss this further via the-team@hal-robotics.com / t.schwartz@hal-robotics.com

Also I am confused, may I ask what is the "create loop" feature you mentioned?
Best, Thibault

DogaKiralioglu, 2024-05-21:
Hi Thibault,

Thank you for the in-depth answer. It is a bummer that the features will not be made public, but I can understand the reasons behind. 

And the "Create a Loop" feature was just something I saw in the Advanced Programming section of the documentation here: https://github.com/HALRobotics/HAL.Documentation/blob/main/docs/Grasshopper/5-Advanced-Programming/Contents.md

Best, Doga

closed 2024-05-21 by thibaultschwartz