HALRobotics/Beta #141: [BUG] missing extensions in the HAL Robotics Framework

Cianelli, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

Hey,
as you can see in the pasted image, I have a problem with finding the extensions in the HAL Robotics Framework Software.
Some time ago I already had installed the extensions successfully, but I recently had a problem with Enscape and HAL, but BEFORE I even did any changes to my system, the extensions were missing. I since then managed the Enscape problem. But the extensions are still missing. I downloaded the newest *.exe from the HAL user portal, re-installed the Plug-in, nothing seemed to bring them back.
Any idea why they disappeared or any solution to get them back?
Thanks in advance!
Yannic
[image]

### Steps To Reproduce

see in the description

sebandraos, 2024-05-28:
Hi @Cianelli ,

This looks like a corrupted installation (possibly linked to Enscape but not sure). Could you please try a [complete uninstall](https://docs.hal-robotics.com/Overview/0-Administration-and-Setup/Contents.html#05-complete-uninstallation) and try again?

Thanks,

Cianelli, 2024-06-04:
Hey, thanks, that worked for me just fine!

Greetings
Yannic

closed 2024-06-04 by sebandraos