HALRobotics/Beta #143: [BUG] HAL Robotics does not load on Grasshopper

ldemt, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

Dear HAL Team,

Since some days (approx 1 week) it happens that the grasshopper loading splash screen takes a lot of time when loading HAL.MCNeel.Grasshopper1.Client. Then when Grasshopper finishes loading, there is no HAL plugin showing up.

If i unplug/replug the robot's ethernet cable and try again 2-3 times it works again, so I guess this might be due to network but i'm not sure.

I have done a clean install of HAL (deleting programdata folder and deleting HAL distribution), up to date and removed other GH plugins.

I can't reproduce this bug on my other PC.

Best regards

Léo

### Steps To Reproduce

1. load grasshopper

sebandraos, 2024-07-03:
Hi @ldemt 

As this is specific to a single PC it sounds like a configuration issue or conflict somewhere so  a few extra details would be useful:

Are there any error messages in the Rhino console/command line?
Is this since an update?
Do you have any issues connecting to the internet through the browser when connected to the robot network?
Does the same happen if you load Grasshopper without the robot network connected?
Does your other PC have two LAN connections?
Does the issue persist if you swap which ethernet port on your PC is connected to the internet and which is to the robot?

Thanks

ldemt, 2024-07-03:
Hi Sebastian, thanks for your help.

- I don't find any error message in the Rhino command line.
- I am not 100% sure but this might be due to a recent update from HAL or Windows or Rhino 7.
- No, i don't have issue connecting to internet while the robot network is connected.
- Now i have tried to load grasshopper without robot network connected, and the splash screen freezes after loading HAL.MCNeel.Grasshopper1.Client for some minutes, then resumes and grasshopper appears without HAL loaded.
- My other PC has a single wifi connection, so no two LAN.
- The issue persists, I have switched the ethernet ports and the network configurations of the two LAN connections, and tried again to start HAL in grasshopper with the two networks connected. the loading freezes again at the HAL.MCNeel.Grasshopper1.Client loading step, and HAL does not load.

sebandraos, 2024-07-03:
Ok, this is very odd. 

Can you please zip your %ProgramData%\HAL folder and send it over? In a reply is fine or via WeTransfer if it's too big.

Hopefully that will allow us to reproduce the issue. If not, then we will try to set up a similar situation with multiple networks and different connectivity to try and resolve this once and for all.

Thanks,

ldemt, 2024-07-03:
here it is : https://we.tl/t-CuvIwtgrQo

thanks

Léo

sebandraos, 2024-07-03:
Hi Léo,
I have just released an update which should fix part of the problem. Can you please try a clean install one more time? It won't fix the network issues but should at least ensure that there are no crashes if you're only on one network (and might fix the rest if we're lucky)

ldemt, 2024-07-04:
Hello Sebastian,

I have done the clean install and started grasshopper. I have unplugged robot network so I am only on internet.
First try, the loading freezes after HAL.MCNeel.Grasshopper1.Client is loaded, hal robotics is loaded but I can't login.
Second try, loading freezes while HAL.MCNeel.Grasshopper1.Client is loaded and hal robotics is not loaded.

Best

Léo

sebandraos, 2024-07-04:
This is so strange, everything is working absolutely fine on all the PCs we're testing. 
Is your other PC still fine? 
Is one of these PCs controlled by an organisation IT system and the other not?
I think the only way to dig into this further is with a call where we can hopefully try some things out and see if we can discover a root cause. Can you drop us an email on the-team@ our usual domain (hal-robotics.com) and we'll set something up.
Thanks