HALRobotics/Beta #144: Digital Input Declaration in Staubli

tristangobin, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

### problem
When you need to use signals, such as Digital Input or Outputs, you need to declare them and they must appear in the .dtx file generate with other staubli files. 
Signals declaration is supported for Output as soon as you use them in the code. (with a set signal component, for example.)
However, there is no component allowing using input signals, which do not allow declaring input signals.
It should be good to export signals if they are supposed to be used elswhere in custom action for exemple

### use case
In my case, I use a ```wait(digitalInput==true)``` in a custom action. I still need the signal to be declared in the dtx file.

### Steps To Reproduce

Declare input signals in a Staubli controller. 
Export code, no input signal are declared in the .dtx file.

sebandraos, 2024-09-18:
This has been fixed internally and will be included in the next update. All signals that are passed into the controller's "Signals" input will now be exported, not just those that are used in an action. I will leave this open until the update goes live (hopefully this week or next)

sebandraos, 2024-09-20:
Released

closed 2024-09-20 by sebandraos