HALRobotics/Beta #145: Installer with incorrect feed path won't start anymore

tristangobin, 2024-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

Once you set a incorrect feed path the installer seems not to be able to start. It means it is impossible to fixe the feed if necessary. 

### Steps To Reproduce

1. Set a incorrect feed in installer
2. close and restart
3. installer start but it will be stuck at loading windows after an error message validation.
[image]
[image]

sebandraos, 2024-09-24:
Can you please send me your Extensions.config file from "%ProgramData%\HAL", privately if required, and I'll try to reproduce the issue here.
Thanks

tristangobin, 2024-09-24:
I emailed you this.

 Thanks

sebandraos, 2024-12-16:
Fixed

closed 2024-12-16 by sebandraos