HALRobotics/Beta #146: [BUG] Synchronization error when using zero-blend motion settings

ldemt, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

Dear HAL Team,

I have an error when I try to set "zero-blend" motions (motion settings where the blend value is set to 0 mm) in a track and robot cell.

When i set one of the motion settings blend to zero, I get the following error from the "Solve" component and i can't run the simulation :
"1. [ERROR] Ambiguous synchronization order defined. The order of synchronization cannot be defined. This may be due to settings being used in different orders in different procedures."

[bug_blend_motion_synchronized.txt](https://github.com/user-attachments/files/18940101/bug_blend_motion_synchronized.txt)

I uploaded an example Grasshopper file as .txt file, to open it please rename it the extension to .gh.

My Rhino version is :
Version 7 SR38
(7.38.24338.17001, 2024-12-03)

Best regards,

Léo

### Steps To Reproduce

1. Rename the attached .txt file to .gh
2. verify the procedure is executed and simulated correctly
3. Switch the output motion settings component of "blend motion" to "zeroblendmotion" component (it will replace the motion settings used in some moves of the example file)
4.

ldemt, 2025-02-24:
Dear all,

here is a workaround (use the comprehensive "Blend" component with a negligible but non-zero blend value) :

[image]

sebandraos, 2025-02-26:
Hi @ldemt ,

This is a known limitation that probably isn't sufficiently documented but I will attempt to describe here.

The way synchronisation works in our software is by matching the _Synchronisation Settings_ between one continuous sequence of motions and another continuous sequence of motions. To ensure unambiguous matching of those sequences we only allow each _Sync Settings_ to be used once per _Procedure_. The issue that you're facing here is that when the _Blends_ are zeroed out, that motion goes from being a single continuous sequence of motions to a series of discrete motions. That triggers our matching algorithm to detect that there are uses of the same _Sync Settings_ in multiple sequences in the same _Procedure_. This also clashes with the fact that your Track's _Motion Settings_ have a default non-zero _Blend_ so it's multiple discrete motions trying to match to a single sequence. Assigning the same blend to the Track won't fix the problem but will give you a more explicit error:
```
[ERROR] Synchronization settings reuse found. Synchronization settings have been found in two distinct sequences of the same procedure.
```

The solution is to generate one _Sync Settings_ per _Motion_ when the _Blends_ are zeroed. You can obviously do that manually but there are potentially a few different ways we can simplify this from a user perspective;
1. We can highlight this as an error during the _Motion_ creation and suggest a solution there.
1. We might be able to add a new version of the _Motion_  component which will, optionally, automatically generate and number the _Sync Settings_ if we know that they won't be a single continuous motion.
1. We can add another component to generate _Sync Settings_ based on a  set of _Targets_ or count (List Length of the _Targets_) and _Motion Settings_.

Would any of those make more sense to you than the others? 

It's worth noting that none of these will fix the fact that the Track and Robot need to both be blended or unblended.